HomeKidd/Homekit-WS2812B-controller #2: cant upload

kefkevm, 2019-11-19:
hey,

i want the bin files uploading to my wemos (esp8266). i used nodeflasherMCU. but it won's work. can someone explain me how to get the bin files on my board?

HomeKidd, 2019-11-19:
Hi! 

All the commands are at the [Wiki page](https://github.com/HomeKidd/Homekit-WS2812B-controller/wiki/Build-Instructions). 
There are tons of [videos]( https://youtu.be/MHrm7axsImI) on the internet about how to install and use esptool.py for flashing ESP8266 boards! 🤓

Esptool is [Espressif’s ](https://github.com/espressif/esptool) **official** tool for flashing ESP8266 boards, basically Arduino using it too for that job!

kefkevm, 2019-11-19:
i used espressif's official tool but i wont work. i tried another .bin file from another project and this wil work. maybe i need another .bin file because i use a wemos. not a esp???

HomeKidd, 2019-11-19:
Could you post an image of the serial what happens when you try to upload it? 
What command do you entered for flashing the wemos?
Wemos is an ESP too so it have to work.

kefkevm, 2019-11-19:
[image]

the files are uploaded the program said. but my iphone dont see the wifi network of it.

HomeKidd, 2019-11-19:
I’ve never seen this program!😳 Thats **NOT** the official Esptool by Espressif!
Please use the method shown in the Wiki or follow the video i’ve linked in the second comment!

Which Wemos do you have? There are some clone ones and these requires QIO mode for flashing!

kefkevm, 2019-11-19:
i downloaded the program here: https://www.espressif.com/en/support/download/other-tools 

my wemos is 8266EX based, i have another wemos. 8266MOD. but it gives the same result. no wifi startup or anything.

kefkevm, 2019-11-20:
with CMD same problem. cant upload the file. maybe i enter the wrong text in the cmd prompt?

I wrote this in my CMD; 

esptool.py -p com4 --baud 115200 write_flash -fs 4MB -fm dio -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 ledstrip.bin

because my wemos is com4 on my pc.

HomeKidd, 2019-11-20:
Please upload picture of the command line after flashing is done!

Also the in the command the rboot.bin / blank_config.bin / ledstrip.bin needs to point to the directory where the files are located ( e.g. C:\your\path\to\the\file\ledstrip.bin)

woainhinhi1314, 2019-11-22:
=================================================================
Can connect to WIFI after refreshing but cannot connect to HomeKit，The following is the serial port display return information   
=================================================================
>>> HomeKit: Starting server
!!! HomeKit: Failed to read flash magic
>>> HomeKit: Formatting flash at 0x100000
!!! HomeKit: Failed to erase flash
>>> HomeKit: Generated new accessory ID: 50:8D:56:7D:DB:CD
!!! HomeKit: Failed to write accessory ID to flash
>>> HomeKit: Generated new accessory key
!!! HomeKit: Failed to write accessory key to flash
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=HomeKid-9C361D-508D md=HomeKid NeoPixelpv=1.0id=50:8D:56:7D:DB:CDc#=1s#=1ff=0sf=1ci=5sh=D+fTpg== Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
!!! HomeKit: Failed to compact data: sector data read error
!!! HomeKit: Failed to write pairing info to flash: max number of pairings
!!! HomeKit: [Client 4] Failed to store pairing (code -2)
>>> HomeKit: [Client 4] Closing client connection

HomeKidd, 2019-11-22:
@woainhinhi1314  Couple minutes ago my friend has the same [issue](https://github.com/maximkulkin/esp-homekit-demo/issues/113).

When flashing the ESP please change the command according to [Wiki page](https://github.com/HomeKidd/Homekit-WS2812B-controller/wiki/Build-Instructions#putting-device-into-flash-mode)

The problem is HomeKit wants to **erase flash sector that not exist when -fs set to 8m _(or 1M)_ like in RavenCore**! Change it to **-fs 32m** OR **-fs 4M** for 4MB flash size and it will solve the problem! 🤓

`--baud 115200 write_flash -fs 32m -fm dio`

woainhinhi1314, 2019-11-23:
=============================================================

@woainhinhi1314  Couple minutes ago my friend has the same issue.
 
When flashing the ESP please change the command according to Wiki page
 
The problem is HomeKit wants to erase flash sector that not exist when -fs set to 8m (or 1M) like in RavenCore! Change it to -fs 32m OR -fs 4M for 4MB flash size and it will solve the problem! 
 
--baud 115200 write_flash -fs 32m -fm dio
  
—

========================================================================

When can I join the ws2812 blinking effect gradient?

kefkevm, 2019-11-24:
no  i dont know how i can point the files to directory location.  i have been bussy last week, this is why my reaction is delayed

kefkevm, 2019-11-24:
[image]
[image]

so far i am

HomeKidd, 2019-11-24:
First things first: the erase command is went well. But the flashing is failed because Python doesnt find the esptool script as shown in the log output.

Second: theres the path to your files: 
`C:\Users\Kevin\Downloads\`

 Try to run the command from where you first run it with the right path to the files: 

C:\Users\Kevin\Downloads\rboot.bin

C:\Users\Kevin\Downloads\blank_config.bin

C:\Users\Kevin\Downloads\ledstrip.bin

[image]

When the command runs as it should be you must see the log output as in Arduino (the Current state of flashing in %). Also if you open up Arduino serial monitor (115200 baud) you should see something in serial after succesfull flashing

[image]

kefkevm, 2019-11-24:
[image]

I have done this, looks right to me but still see nothing in my network list. also arduino IDE dont gives me anything in his monitor.

HomeKidd, 2019-11-24:
What flash size do you have? Because first: esptool doesnt print out the Current flash size, second: the flash params set to 0x1000 and 0x2000 but esptool writes it to 0x00001000 and 0x00002000🙄

Run --flash_size to detect it. Its seems to me that the flash size is more than 4Mb (16Mb like in Wemos D1 Pro)🙄
I’ve seen flash writing to 0x00001000 when i’ve used ESP32 with 16Mb flash

kefkevm, 2019-11-24:
weird, i tried 3 different wemos chips, 1 new wemos and one esp32. but still the same. 0x00001000 always displayed. do i have to change my windows settings maybe?

HomeKidd, 2019-11-25:
> 
> Also change he flash mode from DIO to QIO that maybe works.
>

Only thing you can check is changing flash mode from DIO to QIO. But the 1.2 sec flashing is still too fast. Other thing is to download the bin files again!

kefkevm, 2019-11-25:
i downloaded the bin files again. now it is working!

HomeKidd, 2019-11-25:
> i downloaded the bin files again. now it is working!

It looks like the downloading doesnt finished properly. I was thinking about this too but i'm not mentioned this option 😄 

I'll close the issue finally

closed 2019-11-25 by HomeKidd