HomeKidd/Homekit-WS2812B-controller #7: Wi-Fi from esp8266 not recognized

JCpi, 2020-03-29:
Hi, i followed instructions, but after flashing esp (NodeMCU v3), it's just blinking the led wihout wi-fi, I'm sure it's the firmware problem, because this firmware https://drive.google.com/file/d/1AzWJmgX-aS_xijP5oKopCuhYHjpF2NpJ/view works, but there's another problem...

HomeKidd, 2020-03-29:
Hi!

Which LED is blinking? The WS2812B or the built-in led? Which pin do you connected the data pin of ws2812b? Magic home is a different hardware and code!

JCpi, 2020-03-29:
Thanks for the quick response, I mean the led on the esp itself, I haven't connected the led strip yet, because I can't connect the esp to the Wi-Fi, I also tried to change the Flash parameters, but to no avail, none of my phones see Wi-Fi "HomeKid-"

JCpi, 2020-03-29:
Finally, when I changed the memory size to 16MB, the iPhone saw the wi-fi "HomeKid-", the homekit app sees the esp, but when trying to connect, there's error message saying "Home couldn't connect to this accessory" :(

HomeKidd, 2020-03-29:
> Finally, when I changed the memory size to 16MB, the iPhone saw the wi-fi "HomeKid-", the homekit app sees the esp, but when trying to connect, there's error message saying "Home couldn't connect to this accessory" :(

I've never seen a NodeMCU with 16MB flash before, all of my modules are have 4MByte (aka 32 mBit !!! ) flash size and this is why you can't pair it with your phone: I'm using a specified flash sector to store the HomeKit pairing data, but when you flashing the accessory with flash size other than 4MB, the flash sector will not present. Same issue as #2  😄 Maybe you can try out 32MB settings because 32MB (megabyte) not equal to 32mB (megabit).

I've tried it with my wemos d1 mini and its works like a charm so the code is working, seems to be a hardware issue to me. Please try out other ESP8266 based module that has the right hardware 😄

closed 2020-03-30 by HomeKidd