HomeKidd/Homekit-WS2812B-controller #10: Ap problems

Hyldig01, 2020-04-18:
When I acces the Ap first time, it loads a blank site, I can’t see the networks

I have reset and re flashes the chip. Any ideas?

HomeKidd, 2020-04-19:
Hi! Thanks for comment a new issue! 

I’m also facing the same problem after i’ve updated my devices to the latest iOS 13.4.x! Since that all my other projects (without any code change) producing this problem, so seems to be something changed in iOS with handling Captive Portals😩 Currently the only solution is to wait a little bit more and after a few seconds (in my case 10-20sec) the captive portal’s page will show up🙌

Until this will be patched in [esp-wifi-config](https://github.com/maximkulkin/esp-wifi-config) i’ll leave this issue open! 😉

egorist, 2020-05-01:
From another projects like esp-HomeKit-devices (HAA) I know that this problem was solved in version 1.9.2. I’m not developer, but you can see code yourself and make patch😄

HomeKidd, 2020-05-02:
> From another projects like esp-HomeKit-devices (HAA) I know that this problem was solved in versions 1.9.2 - 1.9.3. I’m not developer, but you can see [Code](https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_1.9.2) yourself and make patch😄

Thanks, I've tried Jose's solution ( aka for testing copied the code from ravencore ) and its better, but still takes 5-10 second to load the UI so i still leave this issue opened.

HomeKidd, 2020-05-02:
**This issue will be patched in the next v2.0.3 release!** 
Seems to be the AP works as before, it takes about 4-5sec to render the UI instead of that 30sec 😄
So I'll close this issue 🙌

closed 2020-05-02 by HomeKidd

HomeKidd, 2020-05-10:
Issue fixed in the latest [v2.1.0 release](https://github.com/HomeKidd/Homekit-WS2812B-controller/releases/tag/2.1.0)👌