HomeKidd/Homekit-WS2812B-controller #12: WS2812B Effects

vglucas, 2020-04-26:
effects 0 - Static - (No flashing. Simply old static light) not working, the strip continues to make effects

HomeKidd, 2020-04-27:
Hi!

What do you mean by not working? Just because if you turn off the LED strip the Effects should be turned off and the effects value automatically sets to 0. I've checked it on my test strip and mine sets back to 0 - Static after i've turned off the strip and also by setting WS2812FX value to 0 😄

HomeKidd, 2020-04-30:
Any information how can i reproduce this bug?

HomeKidd, 2020-05-05:
Since theres no activity and more information about the bug, i close the issue. If the bug still present you can re-open it!🤓

Also i’m still working on WS2812FX so this issue might be gone in future with the newer releases!🙌

closed 2020-05-05 by HomeKidd