HomeKidd/Homekit-WS2812B-controller #15: Flashing to NodeMCU

gio-dude, 2020-07-04:
I think this is a really cool project and would love to give it a shot. I have some nodemcu boards that i would like to use instead of the wemos D1 mini. I tried flashing the software using [this](https://github.com/nodemcu/nodemcu-flasher) flasher. the flasher gives me a green check mark and after a restart blue led blinks but it never shows up on the wifi. 

Are the nodeMCU supported?

HomeKidd, 2020-07-05:
Hi! 
Yes, the nodeMCU is supported because i have a few of them  too 😄 

> it never shows up on the wifi.

Do you mean its not connected to Wi-fi? Or the captive portal never shows up?

If the first one, please read the [Wiki](https://github.com/HomeKidd/Homekit-WS2812B-controller/wiki/HomeKit-Setup#wi-fi-setup)

In case the captive portal doesnt show up, please read Issue #2  and Issue #7 for more info, because Flash Size and Flash Mode params are affect the Captive portal's / HomeKit's behaviour 😄 

Also some log files would be nice whats really happening 😄

gio-dude, 2020-07-06:
yes sorry about the delay. I had some time last night to mess with it. After reading issue #2 and #7 i've learned a lot. At first i thought the node MCU wasn't supported by the official flasher from espressif. after watching the video linked in issue #2 i figured it was my settings. so I switched to the official flasher. I got it to flash successfully but it still wouldn't show up for the captive portal. 

What tried and all gave successful flashes but no captive portal or home kid showing up in wifi. (I did restart the ESP after each flash)
-re download of the bin files.
-I tried 4mb of memory in DIO and QIO mode 
-32mb in DIO and QIO 

So then I read you recommended trying the BME280 project. 
I flashed that and home kid didn't show up but LCM something did and when I clicked it, it loaded the captive portal I Selected my WIFI put in the password and left the OTA alone. when I hit join it started to load then froze. So I restarted it and got to the captive portal agin and now it asked for the wifi but not the OTA. I joined but homekit could find the device when i scanned the QR code. 

I hope my troubleshooting makes sense and really appropriate the help. I not sure how to upload the logs because i don't know how to find any logs.

HomeKidd, 2020-07-07:
Strange issue 😳 using the OTA part is buggy (OTA part is not my code) so sometimes its fails to download the code properly. 
You can se the log files using e.g.: Arduino’s serial monitor. 

Did you erased the flash before upaloding? 

Also did you tried to press and hold the flash button on the NodeMCU to reset the code?

gio-dude, 2020-07-08:
Ok thank you for your help it is much appreciated. I got it to work. I did DIO on 32mb and I erased first and then opened serial monitor in arduino. I not sure which one fixed it arduino or erase that was poor troubleshooting on my part for trying at the same time. 

I have to say the hold the flash button to reset is an amazing feature because I have had it stop responding and had to reset it. 

Great project and I already ordered some BME280 sensors to do your other project.

closed 2020-07-08 by gio-dude