HomeKidd/Homekit-WS2812B-controller #17: Device after some time not reachable anymore

Caleb79, 2020-07-15:
Hi,
after some time the devices are not reachable anymore in Homekit. I have now 4 of these in my Homekit and after a while they are gone. Not every but here and there one of these is not responding so you cannot use them. A restart by powering off and on helps and they are back.
It's also strange that sometime they are not reachable via Wifi anymore although the wifi repeater is really next to them (10cm). But sometimes they are still reachable via Wifi but not in Homekit.
What can I do to not do the restart by powering off these devices?
What is you experience with this behavior, do you have it as well?

gio-dude, 2020-07-18:
I've been having the same issue. ￼ usually homekit will say device not responding. Restarting the device doesn't help me I have to reset and join WiFi and add back to homekit. I'm been trying to figure out what I'm doing wrong because it started happening once I tried powering the device from the gio. glad I'm not the only one seeing this issue. ￼

IonutMindrescu, 2020-09-13:
I'm still searching for a fix but I don't think it's related to this firmware. Looks like Apple Home is kinda crap...(both iOS 13 and iOS 14 Beta 8)

HomeKidd, 2020-10-05:
> I'm @still searching for a fix but I don't think it's related to this firmware.

Yes, sadly thats true😢
I have a Honeywell T6 HomeKit thermostat, so a real HomeKit product not a DIY solution and it gets unreachable very often😬 but i can reach it from the Honeywell app so its on the network and works 😅

The main problem should be Apple’s mDNS requirements and bonjour implementation as i know. Other thing to mention is using multiple APs (Wi-fi access points) also can trick HomeKit accessories: eg.: phone connected to AP 1 and HomeKit accessory connected to other AP (via wi-fi). This is also a well known problem with wi-fi HomeKit accessories 😬 (but since an update to esp-homekit a few months ago this problem was solved for me)
Other thing to mention is ESP8266 not supported MCU for native HomeKit by default. Even Espressif (ESP8266 manufacturer) says only ESP32 supported for official/commercial HomeKit 😁

I know the developer of esp-homekit will change the mDNS part in the future, have to take a look at its solving the issue or not ✌️🤞

Caleb79, 2020-10-05:
Thanks.
Then let's wait for the update.

IonutMindrescu, 2020-10-05:
@HomeKidd any eta for the coming update? Cool thing about your homekit neopixel is that using eve app will resync the status from the homeapp :)) so it’s something related to the thing that the connections are not staying alive. Sometime when I spam 1 unreachable device this will trigger the sync for the rest of the devices too. It’s kinda weird and if the original homekit devices have this issue too then it’s all crap🙄

Partizan7676, 2020-10-19:
mDNS its local network problem. In my case helps setting domain name on router for "local" instead ubnt. Checking of that - see hostname in router leases and try to ping it with full name ex: HAA-xxxx.local if it says cant resolve hostname - check mDNS service on your router. 224.0.0.251 port 5353 protocol UDP - if that blocked by firewall - you get status "no response" for device.

Caleb79, 2020-10-19:
> mDNS its local network problem. In my case helps setting domain name on router for "local" instead ubnt. Checking of that - see hostname in router leases and try to ping it with full name ex: [...]

Can you explain what is needed in a router setting? I'm having exactly that issue. I've a AVM FritzBox (don't know if that is familiar to you).

Partizan7676, 2020-10-19:
@Caleb79 try to ping your device by name. Please tell model number. FritzBox have a lot of different models.

Caleb79, 2020-10-20:
@Partizan7676 I've a FritzBox 7590, newest model with latest firmware.
When I'm doing a ping on this address I got a lot of results. I also tried to ping different types of devices, like my iMac, an iPad etc. Sometimes you get a result, sometimes not. But never when the device is something like an ESP.

Partizan7676, 2020-10-20:
@Caleb79 as i check KB of fritzbox - no information at all... so if you can take for some moment other router - you will see it solves that problem. My model is EdgeMax ER-6p - is good quad core model. Other good one is a UniFi Dream Machine.

Caleb79, 2020-10-20:
@Partizan7676 Interesting device but this actually not applicable to my situation/requirements (wifi, dect).
But knowing that issue I can search for different options (I've seen that there is a kind of special firmware for FritzBox: freetz). Let's see.

HomeKidd, 2020-10-20:
**Because it seems to be a general issue with HomeKit and not only related to this firmware i close the issue✌️**

@Partizan7676  as i know You also have issues with UniFi APs what i’ve mentioned before as the cause of HomeKit “not responding” issue. Apple released iOS14.1 with fixes for UniFi APs. I always kept all my UniFi APs up to date so the issue should be with iPhone/iOS. Hopefully it will solve it🤓

[image]

closed 2020-10-20 by HomeKidd