HomeKidd/Homekit-WS2812B-controller #24: Adaptive Lighting

csupi003, 2020-09-09:
Hello, 

Is it possible to add the new iOS 14's adaptive lighting feature ? 

Thanks in advance,

HomeKidd, 2020-09-09:
Hi!
Currently the Adaptive Lighting part of HomeKit is not open source, only available for MFi licensees.
When the first beta came out i’ve started instantly thinking of adopting that feature 😛If someone fully reverse engineer it i’ll add that feature, but until a (stable official) iOS14 release that doesnt make sense for 99% of users 😉✌️

HomeKidd, 2020-11-12:
Added to my [Todo](https://github.com/HomeKidd/Homekit-WS2812B-controller/wiki/To-Do)✌️

Sadly still nobody knows how [Adaptive Lighting works](https://github.com/ebaauw/homebridge-hue/issues/766) 👷‍♂️

renssies, 2020-11-12:
It has been implemented in the beta HAP-NodeJS (the base of Homebridge). But it's accidentally called "Ambient Lightning": https://github.com/homebridge/HAP-NodeJS/blob/beta/src/accessories/Light-AmbientLightning_accessory.ts

acrogenesis, 2022-09-10:
Hello, I'm just starting with HomeKit/light strips curious if there's been progress regarding Adaptive Lightning.