HomeKidd/Homekit-WS2812B-controller #26: WS2811

ChopsKingsland, 2020-10-18:
Quick question: I'm thinking of buying some neopixels for my Christmas tree later in the year, and I would like them to be compatible with HomeKit. Will it work with these LEDs, even though they're not WS2812B, they're WS2811, https://www.amazon.co.uk/dp/B07TN6MF4G/ref=cm_sw_r_cp_api_i_InjJFbVJESXYR ?

Thanks

HomeKidd, 2020-10-19:
Hi!

I have no experience with WS2811 LEDs because most of NeoPixel LED strips are based on WS2812B.
As i know [WS2811](https://www.suntechleds.com/info/ws2801-vs-ws2811-vs-ws2812-digital-pixel-led-s-36671801.html) is a **12V LED strip** and its just a normal LED strip with  each 3 LEDs grouped together with a WS281x driver IC which means You cannot drive each LED individually, just in group of every 3 pixel as I know. 

The biggest reason why I'm not recommending WS2811 is because _- as i've mentioned -_ its just a normal LED strip with WS281x driver IC. If the manufacturer of these Christmas Lights are changed the RGB layout to GRB or BGR the LEDs will produce off colors. 

BTW you can try it, it should work but you have to keep in mind the 12V power supply 😄

closed 2020-10-19 by HomeKidd

Partizan7676, 2020-10-19:
2811 needs higher signal level. by it's specs NodeMCU gives less than needed by 0.1V

HomeKidd, 2020-10-19:
> 2811 needs higher signal level. by it's specs NodeMCU gives less than needed by 0.1V

My custom PCB and any ESP8266 based board with the **recommended IC** gives 4.5-5V logic high level instead of the 3.3V that has the ESP8266 by default. 
According to WS281x datasheet logic High should be VDD X 0.7V = 3.5V✌️

Partizan7676, 2020-10-19:
thats good enough! when i first time use 2811 - i had that trouble and leds are not working.

ChopsKingsland, 2020-11-15:
So, can I use 5v WS2811 LEDs by powering them from an external power source?

HomeKidd, 2020-11-15:
> So, can I use 5v WS2811 LEDs by powering them from an external power source?

Connect everything as in the Wiki and it should work ✌️ But my hardware only works with 5V LED