HomeKidd/Homekit-WS2812B-controller #29: In my ws2811 green and blue led are schwitched. Could you add a option(maybe in setup to change RGB To RBG)?

RowMax03, 2020-11-23:


HomeKidd, 2020-11-23:
WS2811 strips are not [supported](https://github.com/HomeKidd/Homekit-WS2812B-controller/wiki#supported-led-strips) due to this. WS2811 can be only controlled in group of 3 LEDs and basically these are just simply RGB LED strips with WS28xx IC added to it. Due to the complexity the code, the effects and these type of “errors” only WS2812 and upwards supported only

closed 2020-11-23 by HomeKidd

RowMax03, 2020-11-23:
The effects all work fine on mine and I have them currently installed. Can i acces the source code? I can change it myself then. I normally use .ino and i have no experience with bin

ChopsKingsland, 2020-12-05:
That's what I was thinking... because my WS2811 work, and _aren't_ being controlled in groups of 3 (the colour chase works, 1 LED at a time)