HomeKidd/Homekit-WS2812B-controller #33: Where is the source code ?

toome123, 2020-12-09:
Thanks, for great wiki and this repo at all. But where I can find source code of this bin?

HomeKidd, 2020-12-09:
Theres no source code due to the complexity of my custom version of esp-open-rtos, custom libraries etc. 99% of people was unable to compile the code when it was public.

But the source of the project is available in the readme.md, my code is based on that and its public:

> This project uses the Apple HomeKit accessory server library [ESP-HomeKit](https://github.com/maximkulkin/esp-homekit) from [@MaximKulkin](https://github.com/maximkulkin) for [ESP-OPEN-RTOS](https://github.com/SuperHouse/esp-open-rtos).

closed 2020-12-09 by HomeKidd

dsully, 2021-02-02:
Seems like it'd be worth the time to ensure that the source is able to be compiled with proper setup and/or Docker image with cross-compilers.