HomeKidd/Homekit-WS2812B-controller #34: mdns_recv: pbuf too small

dj-nic, 2021-01-01:
Hey
I just tried to setup the code and add it to HomeKit.
In homekit it says the device couldn't be found... so i looked up in the serial monitor and it keeps saying mdns_recv: pbuf too small
even the wifi is connected and the service is running.
Can someone help me with that?
thanks in advance!

[image]

HomeKidd, 2021-01-04:
Hi!

Which ESP did you use and which settings for flash it?

[Other have this issue](https://github.com/maximkulkin/esp-homekit-demo/issues/11) when the flash size params was incorrect! Please  erase the flash and try again with the correct settings for flash size:
ESP-01 has 512kb -> 4mbit flash size NOT supported
ESP-01S has 1MB -> 8mbit flash size
ESP-07/12E/12F/12S has 4MB -> 32mbit flash size

dj-nic, 2021-01-08:
Hey!

Im using an ESP8266 for this and i will try it out with other flash sizes soon
because im a bit busy in the last time.
Thanks for the response!

dj-nic, 2021-02-05:
OH Sorry I forgot to answer here 😅.   Yes I tried it again to reflash with other options and so on... It crash/restarts at some options and on the others I still get the error with pbuf to small.

HomeKidd, 2021-02-05:
Did you tried it with other Wemos?

dj-nic, 2021-02-05:
No actualy not because it was the last that wasn't in use. But it is Working if you think it is might broken. Right now it's plugged in into my pc with an GRB stripe for Ambilight. I Will buy some new esp8266 and esp 32 (hopefully) on monday.

dj-nic, 2021-02-14:
Well I got it working now with a new chip idunno why the other chip said "no" but some effects are a bit laggy is it maybe because I flashed the chip to 40MHz instead of 80MHz?
[image]

closed 2021-02-27 by HomeKidd