HomeKidd/Homekit-WS2812B-controller #35: [REQUEST] BOM-File for V3 PCB

freumichCH, 2021-01-04:
Hello HomeKid 

First i want to thank you for your awesome work on the WS2812B Homekit solution. 🥇 
It's working very well and i'm so happy with it.

Now i want to go the next mile. i liked to order your PCB-board from BCP-Way. 
I just want to ask you if you can provide me the needed BOM-File for this Project. Unfortunaly I'm not familiar with the different manufactures and models of the parts and i dont want to make a misstake. 

I'm looking forward to read from you and wish yoou a lovely evening. 😊

caiotadashi, 2021-01-04:
hey!

i've requested too and he sent me the BOM and the Pick&Place files

you will need to send these two files

[Homekit-WS2812B-controller files.zip](https://github.com/HomeKidd/Homekit-WS2812B-controller/files/5767082/Homekit-WS2812B-controller.files.zip)

freumichCH, 2021-01-04:
Wow that is exactly what i was looking for.
Thank you very much for your fast respond. 

I will close this Request

closed 2021-01-04 by freumichCH