HomeKidd/Homekit-WS2812B-controller #38: NodeMCU dead after flash

addybc93, 2021-02-05:
Hello, I flashed the ledstrip.bin file to a NodeMCU, it took about 1,5 minutes and after that the board is completly dead. The blue led flashes once when Connected to usb but the board is no longer recognised by the pc. Maybe you can point me to a direction to solve this issue?

HomeKidd, 2021-02-05:
Hi!

Looks like the [NodeMCU is bricked ](https://github.com/esp8266/Arduino/issues/284#issuecomment-103665182) which is a very common issue sadly and  hardware related.

Which settings did you use to flash it?
Did you erased the flash before uploading the firmware?
Do you have any other connection (button, ws2812b)?

HomeKidd, 2021-03-16:
Closed because no activity

closed 2021-03-16 by HomeKidd