HomeKidd/Homekit-WS2812B-controller #41: Support for GRB strips

pnuding, 2021-02-21:
There's a few WS2812b strips and strings out there (and on my balcony) that have their LEDs wired up in GRB order rather than RGB. Without tweaks that means Red is green and vice versa, screwing up color mixing in a number of ways. Any chance to get a GRB version (nevermind a configurable characteristic) or the sources to compile one?

HomeKidd, 2021-02-27:
Hi!

Take a look at issue #29 . This happens with cheaper LEDs, because cheaper ones are just normal RGB LEDs with WS28xx driver IC and its up to the manufacturer which connections they use for RGB channels. This is why i've highly recommended using well known seller like [BTF Lighting](https://github.com/HomeKidd/Homekit-WS2812B-controller/wiki#supported-led-strips), all of my LEDs ordered from them has the correct RGB order 😄 

**BTW i'm working on a solution for changing the RGB on first setup 🤓**

> This project uses the Apple HomeKit accessory server library [ESP-HomeKit](https://github.com/maximkulkin/esp-homekit) from [@MaximKulkin](https://github.com/maximkulkin) for [ESP-OPEN-RTOS](https://github.com/SuperHouse/esp-open-rtos).

lukaszwitkowski, 2021-04-21:
Im need that too:)

lukaszwitkowski, 2021-07-20:
Do you add GRB yet?

Inspiro83, 2022-01-18:
+1 . Another .bin file for GRB would help already.

DigitalElise, 2022-05-28:
+1 for this

htx996, 2023-06-27:
@HomeKidd 8266发射出的wifi信号，密码能否清除掉，改为开放式无密码wifi

rexy666, 2023-12-13:
Hello, could you please add support for GRB?
Or allow us to work in your code to add it? Thank you