HomeKidd/Homekit-WS2812B-controller #42: STL for Case Design & Files for Reflow Stencil

Martin1001, 2021-02-22:
I really like your PCB designs. 

Unfortunately I can't find the data for the PCB and for the stencil to download at PCB Way. Could you make them available for direct download (possibly in Git)?

In order to design a nice case for the board, a 3D model of the board would be really helpful. Could you also make this available in Git?

HomeKidd, 2021-02-27:
Hi!

You can order the stencil when you place the order, just check the stencil box😄
[image]

The STL files are not available because the program I'm using for designing the PCBs cannot export any 3D files, i always design 3D parts using the actual soldered PCB 🤓 but currently i have no time to deal with the enclosure

closed 2021-02-27 by HomeKidd

Martin1001, 2021-03-05:
@HomeKidd Thank you very much for your awnser.

But I forget the stencil in my previous order and I would like to order it at eisler.net in Germany. This is much faster for me. It would be great if you provide the stencil file.