HomeKidd/Homekit-WS2812B-controller #46: Wifi not working

ccnh02, 2021-04-12:
I flashed my esp8266, but i cant find the Wifi AP that it is supposed to generate.

Could anyone help me?

HomeKidd, 2021-04-13:
Hi!
Its a common mistake 
Read issue #10 #2 #7 #15 #38

closed 2021-04-13 by HomeKidd

BullzLabz, 2021-07-15:
Hello,

I used this ([Dev Board](https://a.aliexpress.com/_mLadC5z) for flash the "ledstrip.bin" file to my ESP12-F and I try several times with 3 different ESP-12F modules and I can't see the Wifi AP network !

I've also tried different flash mode "DIO" and "QIO" but same issue and I can't choose the flash size with the NodeMCUFlash tool!

What's doing wrong? I missing steps?

This is the result of my flash's :

```
esptool.py v2.6
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: c4:5b:be:6e:24:f5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.4s
Compressed 507328 bytes to 358349...
Wrote 507328 bytes (358349 compressed) at 0x00000000 in 32.4 seconds (effective 125.5 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.
```

MarkM500, 2021-09-15:
Do you BullzLabz find a solution for this chip? Same issue here.
Thanks