HomeKidd/Homekit-WS2812B-controller #47: Firmware Update: Version 1.1.2

optikfluffel, 2021-05-18:
Hey there, since the source isn't public and I can't send a PR:
Could you please release a new image with the firmware version number bumped to `1.1.2 (500)` so the Eve Apps stops showing me an available firmware update?

Thanks a ton ☺️

HomeKidd, 2021-05-18:
Hi!

Yes, thats drive me crazy too but totally forget to change the code 😂 

The FW update "bug" is caused because i've used real Eve's FW number to make some experiments with Adaptive Lighting and I've forget to change it to a dummy version number 😂 

I'll make a release in a few days 😄 
Thanks for reminding me 😇

closed 2021-05-18 by HomeKidd

optikfluffel, 2021-05-18:
Take your time. And thanks for all your Homekidd thingies :)