HomeKidd/Homekit-WS2812B-controller #52: ESP8266EX Module not work

BullzLabz, 2021-07-22:
Hello,

I recently succeeded in flashing the official **ESP8266 12-F** "Espressif" modules without problem with the official software of Espressif.

Today I tried to flash non-official **ESP8266EX** modules several times (I failed to identify the logo) but after flashing, no "HomeKid-xxx" network appears.

I tried the **"QIO"** and **"DIO"** mode, and also 4Mb or 32Mb but nothing helps, it doesn't work.

To make sure that the modules were not bricked, I tried with the Arduino software to upload a test sketch and all the modules work perfectly, it's very strange.

Here are the results of the modules (in Arduino):
- Chip: **ESP8266EX**
- Crystal = **26MHz**
- Flash size detected = **4Mb**

Could you help me or tell me if these modules are compatible please?

andywaplinger, 2021-09-05:
@HomeKidd I'd also like to know if this could work with the ESP8266EX. I assume that 16MB would allow you to remove the 500 LED limit?

MarkM500, 2021-09-15:
Same issue here like BullzLabz. Is there a solution or an answer to this question? It is a D1 min v3.0.

andywaplinger, 2021-12-20:
@MarkM500 @BullzLabz I finally had some time this past week to work on these for myself. I bought both Wemos D1 minis and ESP-12Fs and have gotten them both to work using my Mac. Hopefully you're both using Macs so this is relevant to you.

What I've found is that I have to first flash the controller using the PyFlasher tool with the `ledstrip.bin` firmware. After that's done, I use the `esptool.py` through Terminal to flash all three .bin files to their correct addresses at once.

You would change `usbserial-1420` to whatever your USB device in PyFlasher is: `esptool.py -p /dev/cu.usbserial-1420 --baud 115200 write_flash -fs 4MB -fm dio -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 ledstrip.bin`

From what I can tell, PyFlasher won't ever work because you can't specify the address for the firmware (to my knowledge), so you can't choose different addresses for the different bin files. That's why I've found success with `esptool.py`.

And for some reason I can't just flash it using only `esptool.py`, I have to flash it with PyFlasher first. This process has been working reliably now for 6 controllers, and I'm planning to flash several more.

Let me know if this helps! Hopefully it does, because it's so nice having HomeKit-native LED controllers. 😃

Inspiro83, 2022-01-18:
At first I was facing the same isssue like BullzLabz following the wiki. But using the instructions above it worked. Thanks a lot!

andywaplinger, 2022-01-18:
Awesome @Inspiro83! I'm really glad that helped.

closed 2023-01-27 by HomeKidd