HomeKidd/Homekit-WS2812B-controller #53: how to connect my esp8266 12f and ws2812b.

172505197, 2021-08-15:
I've flash the firmware and successfully added accessories to HomeKit. But I have some questions about how to connect my nodemcu (esp8266 12f) and ws2812b.

I'm not sure which io connects to which port on the ws2812b. Should I connect gnd-gnd, vin-5v? and which pin should connect to Di to control the light. Can you tell me how to connect all these things when using this firmware? thank you very much.

HomeKidd, 2021-08-15:
Hi!
All info you need is available at the [Wiki page](https://github.com/HomeKidd/Homekit-WS2812B-controller/wiki/PCB-Design)
😉

closed 2021-08-15 by HomeKidd

172505197, 2021-08-16:
I’ve read the wiki and it only says that the output pin is gpio 3, I ve connect the output pin with the di on the light, but it doesn’t work. Are there anything wrong with the other connection ?
[image]

HomeKidd, 2021-08-16:
How many LEDs do you want to drive from the ESP itself?
Because it can make around 7-800mA which might be too low amperage for the WS2812 so it doesn't turn on.
Also are you sure that the used LED strip is WS28xx? Some others looks the same but not supported.
More info [here](https://github.com/HomeKidd/Homekit-WS2812B-controller/wiki#supported-led-strips)

Also the most two important thing:

* WS28xx needs logic level shifter because the ESP has 3.3V pins, the LED strip needs 5V ([more info here](https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices)
* The schematic i've posted powers both the LED strip and ESP from external power supply, not USB (USB is only for flashing the firmware or debug) and also has the logic level shifter

172505197, 2021-08-16:
I try to make it work like this. As you just said, I think there is not enough power supply. There seems to have a 3v pin and a vin pin on the nodemcu, what should I do to make it have enough power supply.
[image]