HomeKidd/Homekit-WS2812B-controller #75: Unstable connectivity

jmgutierrezc1, 2023-09-07:
Great project and build instructions. Having the PCB build with the shared files was fast an easy and the quality is very good.

I have to reset and recooonec to Homekit every 2-3 days. I built three of these and none of them is stable. I guess there is still code imporvement. BTW having and OTA option may help in getting improvement to that point but I still see that option in the to-do list

arkn666l, 2023-09-12:
same issue 2-3 days its necesary reconnect, thanks for this amazing project

andywaplinger, 2023-12-19:
@HomeKidd that’s awesome! Are you targeting an approximate release date?

I performed the HomeKit architecture upgrade last week and now nearly all of my controllers are unstable/not reconnecting even though they’re active on the network.

andywaplinger, 2024-02-22:
@jmgutierrezc1 @arkn666l if you want to move to the ESP32, I just created this LED controller with HomeSpan. So far it's working great with my WS2812Bs.

jmgutierrezc1, 2024-06-08:
> @jmgutierrezc1 @arkn666l if you want to move to the ESP32, I just created this LED controller with HomeSpan. So far it's working great with my WS2812Bs.
> 
> [ESP32 HomeKit LED](https://github.com/andywaplinger/ESP32-HomeKit-LED)
What esp32 model are you using? Can the same pub be used?

jmgutierrezc1, 2024-06-08:
> > @jmgutierrezc1 @arkn666l if you want to move to the ESP32, I just created this LED controller with HomeSpan. So far it's working great with my WS2812Bs.
> > [ESP32 HomeKit LED](https://github.com/andywaplinger/ESP32-HomeKit-LED)
> > What esp32 model are you using? Can the same pub be used?

I meant PCB…

andywaplinger, 2024-06-13:
@jmgutierrezc1 I used these ESP32s https://amzn.to/3KF4oBg

I was able to combine them with my existing HomeKidd-designed boards to take advantage of the power and data components already on that board.

[image]
[image]
[image]
[image]